modm-io/modm-devices #4: make generate-stm32 fails for stm32l412c8t

strongly-typed, 2018-10-14:
```
./stm_generator.py STM32L4
[INFO] dfg.stm.reader: Found devices for prefix 'STM32L4': STM32L412C8T, STM32L412C8U, STM32L412CBT, STM32L412CBU, STM32L412K8T, STM32L412K8U, STM32L412KBT, STM32L412KBU, STM32L412R8I, STM32L412R8T, STM32L412RBI, STM32L412RBT, STM32L412T8Y, STM32L412TBY, STM32L422CBT, STM32L422CBU, STM32L422KBT, STM32L422KBU, STM32L422RBI, STM32L422RBT, STM32L422TBY, STM32L431CBT, STM32L431CBU, STM32L431CBY, STM32L431CCT, STM32L431CCU, STM32L431CCY, STM32L431KBU, STM32L431KCU, STM32L431RBI, STM32L431RBT, STM32L431RBY, STM32L431RCI, STM32L431RCT, STM32L431RCY, STM32L431VCI, STM32L431VCT, STM32L432KBU, STM32L432KCU, STM32L433CBT, STM32L433CBU, STM32L433CBY, STM32L433CCT, STM32L433CCU, STM32L433CCY, STM32L433RBI, STM32L433RBT, STM32L433RBY, STM32L433RCI, STM32L433RCT, STM32L433RCY, STM32L433VCI, STM32L433VCT, STM32L442KCU, STM32L443CCT, STM32L443CCU, STM32L443CCY, STM32L443RCI, STM32L443RCT, STM32L443RCY, STM32L443VCI, STM32L443VCT, STM32L451CCU, STM32L451CEU, STM32L451RCI, STM32L451RCT, STM32L451RCY, STM32L451REI, STM32L451RET, STM32L451REY, STM32L451VCI, STM32L451VCT, STM32L451VEI, STM32L451VET, STM32L452CCU, STM32L452CEU, STM32L452RCI, STM32L452RCT, STM32L452RCY, STM32L452REI, STM32L452RET, STM32L452REY, STM32L452VCI, STM32L452VCT, STM32L452VEI, STM32L452VET, STM32L462CEU, STM32L462REI, STM32L462RET, STM32L462REY, STM32L462VEI, STM32L462VET, STM32L471QEI, STM32L471QGI, STM32L471RET, STM32L471RGT, STM32L471VET, STM32L471VGT, STM32L471ZET, STM32L471ZGT, STM32L475RCT, STM32L475RET, STM32L475RGT, STM32L475VCT, STM32L475VET, STM32L475VGT, STM32L476JEY, STM32L476JGY, STM32L476MEY, STM32L476MGY, STM32L476QEI, STM32L476QGI, STM32L476RCT, STM32L476RET, STM32L476RGT, STM32L476VCT, STM32L476VET, STM32L476VGT, STM32L476ZET, STM32L476ZGT, STM32L485JCY, STM32L485JEY, STM32L486JGY, STM32L486QGI, STM32L486RGT, STM32L486VGT, STM32L486ZGT, STM32L496AEI, STM32L496AGI, STM32L496QEI, STM32L496QGI, STM32L496RET, STM32L496RGT, STM32L496VET, STM32L496VEY, STM32L496VGT, STM32L496VGY, STM32L496ZET, STM32L496ZGT, STM32L4A6AGI, STM32L4A6QGI, STM32L4A6RGT, STM32L4A6VGT, STM32L4A6VGY, STM32L4A6ZGT, STM32L4R5AGI, STM32L4R5AII, STM32L4R5QGI, STM32L4R5QII, STM32L4R5VGT, STM32L4R5VIT, STM32L4R5ZGT, STM32L4R5ZGY, STM32L4R5ZIT, STM32L4R5ZIY, STM32L4R7AII, STM32L4R7VIT, STM32L4R7ZIT, STM32L4R9AGI, STM32L4R9AII, STM32L4R9VGT, STM32L4R9VIT, STM32L4R9ZGJ, STM32L4R9ZGT, STM32L4R9ZGY, STM32L4R9ZIJ, STM32L4R9ZIT, STM32L4R9ZIY, STM32L4S5AII, STM32L4S5QII, STM32L4S5VIT, STM32L4S5ZIT, STM32L4S5ZIY, STM32L4S7AII, STM32L4S7VIT, STM32L4S7ZIT, STM32L4S9AII, STM32L4S9VIT, STM32L4S9ZIJ, STM32L4S9ZIT, STM32L4S9ZIY
[INFO] dfg.stm.reader: Parsing 'stm32l412c8t'
[ERROR] dfg.stm.data: Memory model not found for device 'stm32l412c8t'
make: *** [generate-stm32] Error 1
```

```
git submodule status
 ac6a86c6a81ab316f2e4780c5aa7ab794b9e7fb5 cmsis-header-stm32 (ac6a86c)
```

Same with cmsis-header-stm32 at master.

strongly-typed, 2018-10-14:
[ST](https://www.st.com/en/microcontrollers/stm32l4x2.html?querycriteria=productId=LN1980) is giving a [404](https://www.st.com/content/st_com/en/products/microcontrollers/stm32-32-bit-arm-cortex-mcus/stm32-ultra-low-power-mcus/stm32l4-series/stm32l4x2/stm32l412c8.html) on this chip and all others of the stm32l41 and stm32l42 series.

[This AN](https://www.st.com/content/ccc/resource/technical/document/application_note/group0/06/61/1a/a5/de/a1/4c/5e/DM00350156/files/DM00350156.pdf/jcr:content/translations/en.DM00350156.pdf) mentions these chips and refers to DS12469 and DS12470 which are not publicly available yet.

It is great to have postfix `8` and `B` which have a great visual Hamming distance. `</st-irony>`

I'll assume 8KiB of SRAM2 without better knowledge.

The `stm32l412xx.h` gives a hint:
`#define SRAM2_SIZE            (0x00002000UL) /*!< SRAM2 size (8 KBytes) */`
Why isn't this information used programmatically?

salkinium, 2018-10-14:
> ST is giving a 404 on this chip and all others of the stm32l41 and stm32l42 series.

Yeah, ST leaks a lot of these chips beforehand, I suppose they don't want to have an "internal" only version of CubeMX for their evaluation partners. This also happened with the **F**412.

> Why isn't this information used programmatically?

Lack of trust in ST on my part and difficulty of extracting this information from the headers.
You can give it a go, but when I include manually curated info, it usually means I gave up.

PR #5 (2018-10-14): [dfg] Add STM32L41 and STM32L42 series
Closes #4
closed 2018-10-14 by salkinium