modm-io/modm-devices #37: Incorrect table for Flash wait-state on STM32L4+ devices

CrustyAuklet, 2020-06-13:
I am playing with a new devkit, the [SensorTile.Box](https://www.st.com/en/evaluation-tools/steval-mksbox1v1.html) which has a STM32L4+ processor in it.

This processor is in modm-devices already so i figured I would make some examples for modm! The default CubeMX project sets the clock to 120MHz, but trying to do that in modm causes a compile time failure. I traced it back to the device XML `devices/stm32/stm32l4-r5_r7_r9.xml` line 150.

I manually fixed this in my fork, but since I assume this is pulled from cubeMX somehow that doesn't seem like something I should open a PR for (??) The higher clock speed is the main selling point of L4+ so this seems pretty important 😄.

For reference, here is the table(s) from the CubeMX header `stm32l4xx_jal_flash.h`
```
           Table 1. HCLK clock frequency for other STM32L4 devices
           +-------------------------------------------------------+
           | Latency         |    HCLK clock frequency (MHz)       |
           |                 |-------------------------------------|
           |                 | voltage range 1  | voltage range 2  |
           |                 |      1.2 V       |     1.0 V        |
           |-----------------|------------------|------------------|
           |0WS(1 CPU cycles)|  0 < HCLK <= 16  |  0 < HCLK <= 6   |
           |-----------------|------------------|------------------|
           |1WS(2 CPU cycles)| 16 < HCLK <= 32  |  6 < HCLK <= 12  |
           |-----------------|------------------|------------------|
           |2WS(3 CPU cycles)| 32 < HCLK <= 48  | 12 < HCLK <= 18  |
           |-----------------|------------------|------------------|
           |3WS(4 CPU cycles)| 48 < HCLK <= 64  | 18 < HCLK <= 26  |
           |-----------------|------------------|------------------|
           |4WS(5 CPU cycles)| 64 < HCLK <= 80  | 18 < HCLK <= 26  |
           +-------------------------------------------------------+

           Table 2. HCLK clock frequency for STM32L4+ devices
           +--------------------------------------------------------+
           | Latency         |     HCLK clock frequency (MHz)       |
           |                 |--------------------------------------|
           |                 |  voltage range 1  | voltage range 2  |
           |                 |       1.2 V       |     1.0 V        |
           |-----------------|-------------------|------------------|
           |0WS(1 CPU cycles)|   0 < HCLK <= 20  |  0 < HCLK <= 8   |
           |-----------------|-------------------|------------------|
           |1WS(2 CPU cycles)|  20 < HCLK <= 40  |  8 < HCLK <= 16  |
           |-----------------|-------------------|------------------|
           |2WS(3 CPU cycles)|  40 < HCLK <= 60  | 16 < HCLK <= 26  |
           |-----------------|-------------------|------------------|
           |3WS(4 CPU cycles)|  60 < HCLK <= 80  | 16 < HCLK <= 26  |
           |-----------------|-------------------|------------------|
           |4WS(5 CPU cycles)|  80 < HCLK <= 100 | 16 < HCLK <= 26  |
           |-----------------|-------------------|------------------|
           |5WS(6 CPU cycles)| 100 < HCLK <= 120 | 16 < HCLK <= 26  |
           +--------------------------------------------------------+
```

salkinium, 2020-06-13:
> I manually fixed this in my fork, but since I assume this is pulled from cubeMX somehow that doesn't seem like something I should open a PR for (??)

It's not, the CubeMX data has this data encoded with some complicated boolean conditions that I couldn't reverse engineer.

So, this data is currently manually curated 😢, please [fix it here](https://github.com/modm-io/modm-devices/blob/develop/tools/generator/dfg/stm32/stm.py#L157).
Note that because `L4+` is not a valid family, you have to split it by device name (`R5` etc). I guess ST leaves the technical names to the marketing department, ugh.

salkinium, 2020-06-13:
Please also be aware the the L4+ support in modm has not been extensively tested, since their peripherals looked very similiar if not identical to the L4 family, so I just added them without much thinking. So be extra viligant for subtle bugs, perhaps check the generated source code every once in a while.

salkinium, 2020-09-17:
Let me fix this while I'm working on this anyways:
```
 $ ack VCORE RCC-STM32L4* --A  6
RCC-STM32L43_rcc_v1_0_Modes.xml
1679:    <!--         VCORE Range1   VCORE Range 2
1680-    0 WS (1 CPU cycles)   ? 16             ? 6
1681-    1 WS (2 CPU cycles)   ? 32             ? 12
1682-    2 WS (3 CPU cycles)   ? 48             ? 18
1683-    3 WS (4 CPU cycles)   ? 64             ? 26
1684-    4 WS (5 CPU cycles)   ? 80             ? 26
1685-    -->

RCC-STM32L46_rcc_v1_0_Modes.xml
1738:    <!--         VCORE Range1   VCORE Range 2
1739-    0 WS (1 CPU cycles)   ? 16             ? 6
1740-    1 WS (2 CPU cycles)   ? 32             ? 12
1741-    2 WS (3 CPU cycles)   ? 48             ? 18
1742-    3 WS (4 CPU cycles)   ? 64             ? 26
1743-    4 WS (5 CPU cycles)   ? 80             ? 26
1744-    -->

RCC-STM32L4RS_rcc_v1_0_Modes.xml
2358:    <!--         VCORE Range1   VCORE Range 2
2359-    0 WS (1 CPU cycles)   ? 20             ? 8
2360-    1 WS (2 CPU cycles)   ? 40             ? 16
2361-    2 WS (3 CPU cycles)   ? 60             ? 26
2362-    3 WS (4 CPU cycles)   ? 80             ?
2363-    4 WS (5 CPU cycles)   ? 120             ?
2364-    -->

RCC-STM32L4_rcc_v1_0_Modes.xml
1862:        <!--         VCORE Range1   VCORE Range 2
1863-0 WS (1 CPU cycles)   ? 16             ? 6
1864-1 WS (2 CPU cycles)   ? 32             ? 12
1865-2 WS (3 CPU cycles)   ? 48             ? 18
1866-3 WS (4 CPU cycles)   ? 64             ? 26
1867-4 WS (5 CPU cycles)   ? 80             ? 26
1868--->

RCC-STM32L4x2_rcc_v1_0_Modes.xml
1410:    <!--         VCORE Range1   VCORE Range 2
1411-    0 WS (1 CPU cycles)   ? 16             ? 6
1412-    1 WS (2 CPU cycles)   ? 32             ? 12
1413-    2 WS (3 CPU cycles)   ? 48             ? 18
1414-    3 WS (4 CPU cycles)   ? 64             ? 26
1415-    4 WS (5 CPU cycles)   ? 80             ? 26
1416-    -->

RCC-STM32L4x6_rcc_v1_0_Modes.xml
1953:    <!--         VCORE Range1   VCORE Range 2
1954-    0 WS (1 CPU cycles)   ? 16             ? 6
1955-    1 WS (2 CPU cycles)   ? 32             ? 12
1956-    2 WS (3 CPU cycles)   ? 48             ? 18
1957-    3 WS (4 CPU cycles)   ? 64             ? 26
1958-    4 WS (5 CPU cycles)   ? 80             ? 26
1959-    -->
```

closed 2020-09-17 by salkinium

CrustyAuklet, 2020-09-17:
Thanks for doing this @salkinium, looks good! Sorry for reporting then not having time to get it done 😟.

I just ordered a NUCLEO-L4R5ZI-P so I can give it a shot sometime soon time willing. That SensorTile.box has been nothing but problems so far unfortunately.

salkinium, 2020-09-17:
No worries, it's much easier for me to fix, since I still mostly understand how it works 😛

PR #50 (2020-09-17): Fix STM32L4+ wait states
Fixes #37.

cc @CrustyAuklet