modm-io/modm-devices #53: STM32F76x devices have wrong memories

mikewolfram, 2020-09-28:
Hunting down my issue with memory corruptions I figured out that the XML of the STM32F76x devices have wrong entries for the memories:
```xml
      <memory device-size="g" name="flash" access="rx" start="0x8000000" size="1048576"/>
      <memory device-size="i" name="flash" access="rx" start="0x8000000" size="2097152"/>
      <memory name="dtcm" access="rwx" start="0x20000000" size="65536"/>
      <memory name="sram1" access="rwx" start="0x20010000" size="376832"/>
      <memory name="sram2" access="rwx" start="0x2006C000" size="16384"/>
      <memory name="backup" access="rwx" start="0x40024000" size="4096"/>
```
But the DTCM is actually 128kB, which would shift the addresses of SRAM1/SRAM2 too.

If you could point me to the right location to fix it...

mikewolfram, 2020-09-28:
Found a place, in tools/generator/dfg/stm32/stm.py:
```python
    'f7': {
        'start': {
            'flash': 0x08000000,
            'dtcm': 0x20000000,
            'itcm': 0x00000000,
            'sram': 0x20010000,
            'backup': 0x40024000
        },
        'model': [
            {
                'name': ['22', '32', '23', '30', '33', '45', '46', '50', '56', '65', '67', '68', '69', '77', '78', '79'],
                'memories': {'flash': 0, 'itcm': 16*1024, 'dtcm': 64*1024, 'sram1': 0, 'sram2': 16*1024, 'backup': 4*1024}
            }
        ]
    },
```
Is that the right location?

According to STMCUFinder the F76x and F77x have 128kB, all other F7 only 64kB DTCM.

salkinium, 2020-09-29:
Yes, that's the right location. I'm sorry for the bugs, but this info isn't part of CubeMX, so I had to add it manually. This is the reason I really don't like it.

mikewolfram, 2020-09-29:
No worries, nobody's perfect.

Changing the size of the DTCM would that automatically adjust the start address of SRAM?

salkinium, 2020-09-29:
No I don't think so. The algorithm is at the bottom of that file, it may require an extension to allow for custom start addresses per table entry.

salkinium, 2020-09-29:
Ugh, it's a mess:
[image]

mikewolfram, 2020-09-29:
The pic is a bit misleading. The data from STMCUFinder looks more clear to know which F7 has which RAMs.

PR #54 (2020-09-29): [dfg] Fix STM32F7 DTCM sizes
Fixes #53.

cc @mikewolfram 

You can do this in the modm submodule to test this:
```
git remote add salkinium https://github.com/salkinium/modm-devices.git
git fetch
git checkout salkinium/fix/stm32f7_dtcm
```
closed 2020-09-29 by salkinium