modm-io/modm-devices #74: STM32 DMAMUX request mapping data is broken

chris-durand, 2021-10-30:
We are parsing the DMA IP file in the CubeMX data to obtain the mappings between request id and peripherals/signals:
Raw data example:
```
        <RefParameter Comment="Request number" DefaultValue="" Group="" Name="Request" Type="list" Visible="false">
                <PossibleValue Comment="MEMTOMEM" Value="DMA_REQUEST_MEM2MEM"/>
                <PossibleValue Comment="DMA_GENERATOR0" Value="DMA_REQUEST_GENERATOR0"/>
                <PossibleValue Comment="DMA_GENERATOR1" Value="DMA_REQUEST_GENERATOR1"/>
                <PossibleValue Comment="DMA_GENERATOR2" Value="DMA_REQUEST_GENERATOR2"/>
                <PossibleValue Comment="DMA_GENERATOR3" Value="DMA_REQUEST_GENERATOR3"/>
                <PossibleValue Comment="DMA_GENERATOR4" Value="DMA_REQUEST_GENERATOR4"/>
                <PossibleValue Comment="DMA_GENERATOR5" Value="DMA_REQUEST_GENERATOR5"/>
                <PossibleValue Comment="DMA_GENERATOR6" Value="DMA_REQUEST_GENERATOR6"/>
                <PossibleValue Comment="DMA_GENERATOR7" Value="DMA_REQUEST_GENERATOR7"/>
                <PossibleValue Comment="ADC1" Value="DMA_REQUEST_ADC1"/>
                <PossibleValue Comment="ADC2" Value="DMA_REQUEST_ADC2"/>
                <PossibleValue Comment="TIM1_CH1" Value="DMA_REQUEST_TIM1_CH1"/>
                <PossibleValue Comment="TIM1_CH2" Value="DMA_REQUEST_TIM1_CH2"/>
                <PossibleValue Comment="TIM1_CH3" Value="DMA_REQUEST_TIM1_CH3"/>
[…]
```
We are making the assumption that the index in the list represents the request index. Thus, the data is assumed to be ordered by request index and must not have gaps with reserved ids. By accident this seems to be true for G4 but not for H7.

The values refer to constants [inside the CubeHAL headers](https://github.com/STMicroelectronics/STM32CubeH7/blob/ccb11556044540590ca6e45056e6b65cdca2deb2/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_dma.h#L218). At least there is only one such table per device family. We have to somehow copy the data from the header and use it in the generator for finding the correct mappings.

@salkinium

chris-durand, 2021-10-30:
Excerpt from H72x/H73x reference manual:
[image]

Does not match device file:
```
[...]
<request position="123">
	<signal driver="tim" instance="23" name="up"/>
</request>
<request position="124">
	<signal driver="tim" instance="24" name="ch1"/>
</request>
<request position="125">
	<signal driver="tim" instance="24" name="ch2"/>
</request>
<request position="126">
	<signal driver="tim" instance="24" name="ch3"/>
</request>
<request position="127">
	<signal driver="tim" instance="24" name="ch4"/>
</request>
<request position="128">
	<signal driver="tim" instance="24" name="trig"/>
</request>
<request position="129">
	<signal driver="tim" instance="24" name="up"/>
</request>
```

salkinium, 2021-10-30:
It's the same issue I had with the vector table, thus I had to parse it out of the CMSIS header files.
I'm happy to integrate more data sources like the CubeHAL headers, analog to the CMSIS headers.
(You can use the SVN interface to clone only a single path out of a repo).

Note that you will have to use GCC to resolve the macros into a Python dictionary, but [there's already code for that](https://github.com/modm-io/modm-devices/blob/develop/tools/generator/dfg/stm32/stm_header.py#L127-L164).

chris-durand, 2021-10-30:
This thing turns out to be quite messy. For H7 it's straight forward. You can basically parse the information with 5 lines of python using regex. For L4(+) ST decided to put in a condition like `#if defined (STM32L4P5xx)` depending on the specific device. 

@salkinium It all looks like a bit too much effort to give the ST HAL header a proper build environment with including the device specific headers, all dependent ST HAL header and then run the compiler, etc.. Especially if the number of cases per device family is a number like 2 or 3. I feel I should rather do some quick and dirty solution ...

salkinium, 2021-10-30:
Oh, that's sad. The hack it and put it into our other manual data?

chris-durand, 2021-10-30:
I guess I could go for a hacky regex parser that has hard-coded handling for the `#ifdef` section in the L4 HAL. If it encounters unexpected `#if`s it will abort. This is at least better than having to redo lots of manual data extraction once ST updates the headers. I'll give it a try later.

chris-durand, 2021-10-31:
I could extract a dictionary from the CubeHal headers that maps the identifiers in the IP xml file to the actual request id using [that script](https://gist.github.com/chris-durand/69741d902a44424b080f58888ae57fce). It's a bit hacky but we do not really need to understand the data fully. There is only one case where the header contains ambiguous mappings for the same request name to different request numbers. Handling for that has to be hard-coded. If we make sure the script fails whenever that happens we should be good.

chris-durand, 2021-10-31:
I'll add a repo with the cube hal drivers to modm-ext and include them here as a submodule.

salkinium, 2021-12-31:
Has been fixed with #75.

closed 2021-12-31 by salkinium