modm-io/modm-devices #76: Add STM32 peripheral RCC enable bit enums

benedekkupper, 2021-11-03:
One of the most important thing for any peripheral driver is to be able to enable the root clock for the peripheral. On STM32's this is done via RCC A[H/P]nBENR registers. It would be nice if the database contained an enum like in https://github.com/IntergatedCircuits/STM32_XPD/blob/master/STM32F0_XPD/inc/xpd_rcc.h#L78, where the position consists of a register offset from the lowest A[H/P]nBENR register, and a bit offset within the register. Then the appropriate enum values should be bound to the peripheral instances, allowing a peripheral driver to call a uniform RCC driver API to enable/reset/enable in sleep mode a peripheral.

salkinium, 2021-11-03:
You can derive this from parsing the CMSIS-SVD files. [modm parses the CMSIS header files for STM32](https://github.com/modm-io/modm/blob/develop/ext/st/module.lb#L65) (since the SVD files are less accurate) and [uses that to provide such an API](https://github.com/modm-io/modm/blob/develop/src/modm/platform/clock/stm32/module.lb#L112). However, some manual fixing is still required as the mapping is a little fuzzy.

salkinium, 2021-11-03:
We've integrated bit fields before for the [STM32F1 AFIO remap groups](https://github.com/modm-io/modm-devices/blob/develop/tools/generator/dfg/stm32/stm.py#L36), so moving the RCC bit fields into here would probably be useful.

benedekkupper, 2021-11-03:
Yes, so basically what I'm asking is the same as what's [already suggested in the modm parser/generator](https://github.com/modm-io/modm/blob/79eb43b07c6d64d97c1a281b583c8e2fb3cdf7d2/src/modm/platform/clock/stm32/module.lb#L56).

salkinium, 2021-11-04:
Just FYI: I'm currently writing my master's thesis so I don't have the head space to refactor this. I'm happy to review pull requests though, if you want to give it a shot.

There is an effort by me to add a much better, proper Python API to this repo, but it hasn't been a big enough priority for me to finish it: https://github.com/modm-io/modm-devices/pull/35