modm-io/modm-devices #93: STM32G4 Internal temperature sensor has wrong channel mapping

ser-plu, 2022-10-17:
The [reference manual p.615](https://www.st.com/resource/en/reference_manual/dm00355726-stm32g4-series-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf) says:

> The internal temperature sensor (VTS) is connected to **ADC1_INP16** and ADC5_INP4

The _adc.hpp_ file, generated by modm assigns temperature sensor to Channel 17. Similarly, the voltage reference channel is shifted by 1.
```
class Adc1
{
...
	/// Channels, which can be used with this ADC.
	enum class Channel : uint8_t	// TODO: What is the best type?
	{
		...

		Temperature = 17,
		BatDiv3     = 18,
	};
};
```

chris-durand, 2022-10-17:
Hi @ser-plu,

we are aware of this issue, see https://github.com/modm-io/modm/issues/802. The underlying root cause of this is that we are lacking machine-readable data of the ADC channel mappings. The same driver is used across different STM32 series and the channels are most likely only fully correct on the F3 devices the driver has been originally implemented for.

@salkinium Do you have any data that could help to generate this correctly? Otherwise just extracting it manually is the only option right now. This problem also has been bugging me in some actual applications and I had to apply ugly workarounds.

salkinium, 2022-10-17:
I don't have a quick fix for this either, the techdoc parsing code is going to take some time to get finished.

chris-durand, 2022-10-17:
> Maybe you could add a comment line at least at the Channel mapping enum, that it is valid only for F3 devices?

I hope that will fix it: https://github.com/modm-io/modm/pull/921

closed 2022-10-18 by salkinium