modm-io/modm-devices #97: SAM device file inconsistency for MCAN peripheral

rleh, 2023-02-10:
The following lines do not math disjoint sets, because both match e.g. the `same70q19b-an` device, which causes errors in lbuild:

https://github.com/modm-io/modm-devices/blob/e093fe831fe17f1fec54bb31e038e499c22c2830/devices/sam/same70_s70_v70_v71.xml#L265-L266

rleh, 2023-02-10:
I'll take a look at the device file generator...

salkinium, 2023-02-10:
There are bugs in the merging code, I already broke my brain before trying to understand it.

salkinium, 2023-02-10:
So there is a brute-force way to check the merge algorithm by adding the `--check-merge` flag to the generator.
So `./sam_generator.py same7 sams7 samv7 --check-merge` saves each individual tree, then merges, then for every device unmerges the tree and compares it to the seperately saved tree.
Running that does not find any differences though.

salkinium, 2023-02-10:
So that brute-force check is useless, it's clearly wrong, cos the single tree for the `same70q19b-an` is correct:
```xml
    <driver name="mcan" type="sam">
      <instance value="0"/>
      <instance value="1"/>
    </driver>
```

salkinium, 2023-02-11:
I don't have a quick solution for this and we don't even have a functioning test for this to detect.
I would suggest splitting the offending device file using the groups as a workaround that doesn't involve manual patching.

salkinium, 2023-02-11:
Longer term, I would like to actually have a schema that validates such assumptions (no duplicate peripheral instances) so that we can actually test our data.
At the moment that is all implicit and therefore terrible.