modm-io/modm-devices #104: STM32H7 FDCAN type

NorbertToth, 2023-08-31:
In stm32h7-43_53.xml line 209 FDCAN type is stm32-h7, but lbuild not found it.
If I change to stm32, it will be found but the modm FDCAN template driver is not compatible with the STM32H7.

rleh, 2023-08-31:
Hello!
This is not an issue with modm-devices, modm-devices correctly indicates the FDCAN peripheral on STM32H7 series needs it's own driver (`stm32-h7`), which is not (yet) implemented in [modm](https://github.com/modm-io/modm).

This is also indicated in the [peripheral support matrix](https://modm.io/#microcontrollers):
[image]

The peripheral is actually pretty similar to the FDCAN peripheral im STM32G0/G4/U5/... devices with [this driver](https://github.com/modm-io/modm/tree/d37fa244b5a3c5db34889130294e2fd19ecbf990/src/modm/platform/can/stm32-fdcan), but hat configurable message ram like the (also similar) [Atmel/Microchip SAM MCAN driver](https://github.com/modm-io/modm/tree/d37fa244b5a3c5db34889130294e2fd19ecbf990/src/modm/platform/can/sam-mcan).

NorbertToth, 2023-08-31:
Hi, ok thank you